url-kaist/patchwork-plusplus #1: patchwork_parameters

whuzs, 2022-07-28:
hi,  enable_RNR and enable_RVPF is set false by default, Did you not use these two methods?

seungjae24, 2022-07-29:
Hi, thank you for asking the question!
For both methods, I used both of them as stated in the paper. However, as you said, the both parameters are set false by default in the master branch now. This is because it needs additional codes for the general application ( e.g. point types(XYZ, XYZI, etc), sorting order of points ). The original codes, which I have used for experiments, have some dependencies on the data formats. Also, as I apply pybind, there's some changes in the codes compared to the original one and I'm now checking the potential issue in the new code, including R-VPF, now. Please wait for a while, I'm planning to update the generalized code which can execute Patchwork++ for cases as many as possible. Thank you.

Shaoqing26, 2022-08-31:
> 

Hi,
 Is there any update or the planning finish data?

seungjae24, 2022-09-05:
@whuzs, @Shaoqing26 

At first, sorry for the late update of the code.

I have updated the code, which makes 
- **Reflected Noise Removal(RNR)** be available regardless of sorted order of points.
  - The previous code requires the sorted point cloud input in the descending order of ring information. However, the newely updated code can be operated with the unsorted point cloud. 
  - The default value of following parameter `enable_RNR` has been changed to `true`.
  - As the new code doesn't require the ring information, the specific thresholds for RNR have been changed slightly from the previous one: `pc_num_channel`, `noise_filter_channel_num` and `intensity_thr`. The new parameters are `RNR_ver_angle_thr` and `RNR_intensity_thr`. But the essential role of RNR has not been changed.
 
- **Region-wise Vertical Plane Fitting(R-VPF)** be corrected.
  - As I commented before, I have been struggled to correct the functionality of R-VPF due to the data formats and pybind.
  - The default value of following parameter `enable_RVPF` has been changed to `true`.
  - The updated code works properly now. If you find anything wrong, please let me know.

I will close the issue because the related codes are updated.

Thank you :)

closed 2022-09-05 by seungjae24

Shaoqing26, 2022-09-06:
> @whuzs, @Shaoqing26
> 
> At first, sorry for the late update of the code.
> 
> I have updated the code, which makes
> 
> * **Reflected Noise Removal(RNR)** be available regardless of sorted order of points.
>   
>   * The previous code requires the sorted point cloud input in the descending order of ring information. However, the newely updated code can be operated with the unsorted point cloud.
>   * The default value of following parameter `enable_RNR` has been changed to `true`.
>   * As the new code doesn't require the ring information, the specific thresholds for RNR have been changed slightly from the previous one: `pc_num_channel`, `noise_filter_channel_num` and `intensity_thr`. The new parameters are `RNR_ver_angle_thr` and `RNR_intensity_thr`. But the essential role of RNR isn't changed.
> * **Region-wise Vertical Plane Fitting(R-VPF)** be corrected.
>   
>   * As I commented before, I have been struggled to correct the functionality of R-VPF due to the data formats and pybind.
>   * The default value of following parameter `enable_RVPF` has been changed to `true`.
>   * The updated code works properly now. If you find anything wrong, please let me know.
> 
> I will close the issue because the related codes are updated.
> 
> Thank you :)

Thanks for your kind reply.
And I found that this work support ROS, Is this problems process in [ROS Project](https://github.com/url-kaist/patchwork-plusplus-ros)?

seungjae24, 2022-09-07:
Yes. I will also update the corresponding codes in the ROS repository as soon as possible.
Thanks for your interest :+1:.

Shaoqing26, 2022-09-07:
> Yes. I will also update the corresponding codes in the [ROS repository](https://github.com/url-kaist/patchwork-plusplus-ros) as soon as possible. Thanks for your interest +1.

Thanks, Please info me if update it at your convenience. It's an exciting Job. Thanks.

seungjae24, 2022-09-13:
The ROS repository is also updated. You can visit [here](https://github.com/url-kaist/patchwork-plusplus-ros) to try Patchwork++ based on ROS. Thank you :)