url-kaist/patchwork-plusplus #2: What about System Requirements?

zwqnju, 2022-08-08:
Does this repo support running in Ubuntu 16.04? 

As my computer is Ubuntu 16.04, I install an older version of Open3D (0.9.0). But I cannot build Patchwork then.

Or can I run Patchwork on MacOS?

seungjae24, 2022-08-08:
Thank you for the question! Well, unfortunately, I'm sorry to let you know that it has been tested only in Ubuntu 18.04.

However, I can recommend you to build the Patchwork++ without Open3D if you struggle for the Open3D version issue.
Since the Open3D package is only necessary for the visualization of ground and non-ground point cloud, it is not an essential package for the main source code of Patchwork++. You can find out another visualization tool if you want to visualize the result.

Which language do you want to build Patchwork++? C++ or Python? If you let me know, I can help you building Patchwork++ without the installation of Open3D.

zwqnju, 2022-08-09:
Thanks! I have run it in MacOS successfully.

closed 2022-08-09 by zwqnju

zwqnju, 2022-08-12:
BTW, I have built a Docker image successfully, which makes me run your code in my system of Ubuntu 16.04.

YangkoC, 2022-09-13:
How to build Patchwork++ without the installation of Open3D.