url-kaist/patchwork-plusplus #3: Have you tried Patchwork++ on nuScenes?

zwqnju, 2022-08-24:
I run your program on one scene in nuScenes, but the performance is not so good as SemanticKITTI. Is it because nuScenes uses a 32-beam LiDAR and the point cloud is sparser? Do you have any suggestions to improve it?
[image]

seungjae24, 2022-08-24:
Well, at first, I need to say that it is very difficult for me to say which one directly occurs such poor result because the picture is not enough to find out the problem. However, I guess that the sparsity of point cloud due to less beams won't let such result since the false positive points(red points) in your picture are dense enough to be estimated as a proper ground plane in my thought. If the divided patch is far from the sensor origin, the sparsity can be a severe factor which results in a poor plane estimation. I can suggest you to check the uprightness, elevation, and flatness values of those wrongly-estimated planes and compares that values to the thresholds of them, respectively.

closed 2022-09-14 by seungjae24

zwqnju, 2022-10-18:
> 

No..