url-kaist/patchwork-plusplus #5: About patchwork++ parameters

Acuno41, 2022-08-31:
Greetings,

First of all thanks for the great work.

I wanted to try patchwork++ in a custom Lidar datas that i collect from my own sensor and it seems the parameters of the algorithm should be adjusted for my environment. 

Some of the parameters mentioned in your paper but I couldn't understand some of the parameters in the code.

Could you briefly explain the parameters and how it effects the results ?

Best regards.

[image]

LimHyungTae, 2022-09-02:
What kinds of parameters make you confused? Please let me know. I'll explain them :)

Acuno41, 2022-09-02:
Thanks for the response,

1. `num_min_pts = 10; `
why do the algorithm accepts sectors with lower size than this parameter as non-ground ? Isn't it miss classify the ground points in long distance ?

2. `max_range = 80;`
`min_range = 2.7;`
`num_rings_each_zone = {2, 4, 4, 4};`   
Does adjusting these parameters changes the sector sizes ? and How should sector sizes be chosen ?

3. `sensor_height = 1.723;`
If we calibrate multiple Lidars relative to base_link, should we set this height based on the z distance between ground points and base_link distance?  

4. Which parameters should I look-up for behavior on slopes ?
Are they `th_seeds=0.125` and `th_dist=0.125` ?

LimHyungTae, 2022-09-02:
1. Your saying is right, but it's a subtle thing. We felt it does not heavily affect the quality of ground segmentation. If you avoid the situation where these scattered points remain in the non-ground, then reject them all.
2. It is lots of heuristics. Further information can be found in [version 1 paper](https://arxiv.org/abs/2108.05560) 
3. Yes, it is w.r.t. the height of the current frame 
4. `th_seeds`: param. for initialization, `th_dist`: param. for estimation. Thus, if you take the slopes into account, both parameters should be set to be larger.

LimHyungTae, 2022-09-04:
I think all issues are resolved

closed 2022-09-04 by LimHyungTae

Acuno41, 2022-09-05:
Thanks for your suppport

youwyu, 2024-12-25:
Could you please give some estimation or rough equations for th_seeds and th_dist for elevated ground (20 degrees)? Do we also need tune th_seeds_v and th_dist_v?