url-kaist/patchwork-plusplus #8: How to implement patchwork++ in Windows ？

YangkoC, 2022-09-09:
Dear author, I am a new graduate student studying programming. Your algorithm is very novel and excellent. So I want to implement it in my computer environment. Can you provide a tutorial on the implementation of this algorithm in the windows environment? When I use cmake GUI, the error it prompts me is: eigen3_ I really don't know how to deal with dir-notfound, because your algorithm has just come out, and there are few tutorials on the network, which is very difficult for beginners like me, so I ask you for advice. Finally, I wish you every success in your work

LimHyungTae, 2022-09-12:
Hello @YangkoC, in fact, there are no reasons to fail in compiling in Window environment (I guess). I think your error message is about the Eigen library. Did you install it successfully? Please refer to this site: https://eigen.tuxfamily.org/index.php?title=Main_Page

YangkoC, 2022-09-13:
I have installed the eigen library. Now the problem is with open3d. How can I successfully compile your patchwork + + algorithm without using open3d .

LimHyungTae, 2022-09-13:
Is there any particular reason that you don't install open3d? Why don't you just install the library?
http://www.open3d.org/docs/release/getting_started.html

seungjae24, 2022-09-14:
I think this issue has been solved as in #10.

closed 2022-09-14 by seungjae24