url-kaist/patchwork-plusplus #10: How to build patchwork++ without using open3d？

YangkoC, 2022-09-13:
It is too difficult to install open3d Library in Windows environment. I want to know how to build your algorithm without open3d! I hope you can help me, thank you!

seungjae24, 2022-09-14:
Hi @YangkoC, which language will you use to operate Patchwork++? Python or C++?

YangkoC, 2022-09-14:
python

seungjae24, 2022-09-14:
I will open a branch, [10-how-to-build-patchwork++-without-using-open3d](https://github.com/url-kaist/patchwork-plusplus/tree/10-how-to-build-patchwork%2B%2B-without-using-open3d), for a week. It doesn't have any dependency on Open3D. Also, it only works on Python. But, you must be aware that you cannot visualize the ground segmentation result in this branch.

By the way, I suggest you try to install Open3D in Windows and build the version updated on the master. I find that Open3D supports Windows10 environment as written in [this link](http://www.open3d.org/docs/release/getting_started.html).

YangkoC, 2022-09-14:
OK, thank you very much for your help! I will try to install Open3D again

closed 2022-09-14 by seungjae24