url-kaist/patchwork-plusplus #11: Output Indices

qsisi, 2022-09-15:
Hello! Thanks for open-sourcing this amazing algorithm.

I was wondering can the algorithm output the indices of ground (or nonground) with respect to the original input point cloud. This is because the algorithm can only output XYZ point cloud, but when we have attributes more than XYZ, the output indices could be used for indexing the original input point cloud to get the desired output point cloud.

Thanks for your help in advance.

seungjae24, 2022-09-16:
Hi @qsisi. Thank you for having interest on our work.

I can suggest you to add an index information into the structure of PointXYZ defined in the patchworkpp.h file.
Then, insert the index information into the point when points are 
- divided into concentric zone model at pc2czm() function 
- or segmented as non-ground points directly at pc2czm() function or reflected_noise_removal() function.

After the ground segmentation, you can extract the indices of each class as each point has the original index.

qsisi, 2022-09-17:
Thanks for your reply, I will try it out myself.

closed 2022-09-17 by qsisi

tpzou, 2022-10-02:
> Thanks for your reply, I will try it out myself.

Hi，@qsisi , would you mind sending me a copy?  I'm a newbie of C++， this kind of modification is a little difficult for me。

vebjornjr, 2022-10-17:
Did you have any success with this, @qsisi? I have the same needs, we want to keep the intensity information for each point.

qsisi, 2022-10-17:
> Did you have any success with this, @qsisi? I have the same needs, we want to keep the intensity information for each point. @seungjae24

I made some minor modifications based on Patchwork++, which you can try it out [here](https://github.com/qsisi/LidarGroundSeg)

Differences:
1. adding "index" value to the PointXYZ strcuture. ([line 20](https://github.com/qsisi/LidarGroundSeg/blob/6d2bd91317696b8ac4aa15902db16610353f63d8/include/patchworkpp.h#L20) at [patchwork.h](https://github.com/qsisi/LidarGroundSeg/blob/master/include/patchworkpp.h))
2. modify line 585, 589, 593, 597, 601, 376 at [patchwork.cpp](https://github.com/qsisi/LidarGroundSeg/blob/master/include/patchworkpp.cpp)
3. add two member functions getGroundIndices() and getNongroundIndices() at line 607 - line 627 at [patchwork.cpp](https://github.com/qsisi/LidarGroundSeg/blob/master/include/patchworkpp.cpp).

The other part of the code is kept the same as the official patchwork++.

seungjae24, 2022-10-18:
@qsisi, Thank you for sharing the code provides additional functionality :)

vebjornjr, 2022-10-18:
Thank you, @qsisi. I will take a look at it. Have you considered creating a pull request for getting that functionality into this repo too?

qsisi, 2022-10-19:
> Thank you, @qsisi. I will take a look at it. Have you considered creating a pull request for getting that functionality into this repo too?

No :) my implementation is too simple, which I think it's not consistent with the coding standard of this repo. You can try a different implementation, or it's better for the author to add that functionality himself. @seungjae24