url-kaist/patchwork-plusplus #12: offset of sequential pointcloud?

kang-1-2-3, 2022-10-01:
I modify the demo_visualize.py as followings and test it on kitti odometry
`import os
import sys
import open3d as o3d
import numpy as np
# import pypatchworkpp

cur_dir = os.path.dirname(os.path.abspath(__file__))
base_dir = '/media/kang/My Passport/Dataset/KITTI/dataset/sequences'
sequence_list = ["00", "01", "02", "03", "04", "05", "06", "07", "08", "09", "10"]

try:
    patchwork_module_path = os.path.join(cur_dir, "../../build/python_wrapper")
    sys.path.insert(0, patchwork_module_path)
    import pypatchworkpp
except ImportError:
    print("Cannot find pypatchworkpp!")
    exit(1)

def read_bin(bin_path):
    scan = np.fromfile(bin_path, dtype=np.float32)
    scan = scan.reshape((-1, 4))

    return scan

if __name__ == "__main__":

    # Patchwork++ initialization
    params = pypatchworkpp.Parameters()
    params.verbose = True

    PatchworkPLUSPLUS = pypatchworkpp.patchworkpp(params)
    for i in range(len(sequence_list)):
        velodyne_path = os.path.join(base_dir, sequence_list[i], 'velodyne')
        pc_list = os.listdir(velodyne_path)
        if not os.path.exists(os.path.join(base_dir, sequence_list[i], 'nonground')):
            os.mkdir(os.path.join(base_dir, sequence_list[i], 'nonground'))
        for j in range(len(pc_list)):
            # Load point cloud
            pointcloud = read_bin(os.path.join(velodyne_path, pc_list[j]))

            # Estimate Ground
            PatchworkPLUSPLUS.estimateGround(pointcloud)

            # Get Ground and Nonground
            # ground = PatchworkPLUSPLUS.getGround()
            nonground   = PatchworkPLUSPLUS.getNonground()
            time_taken = PatchworkPLUSPLUS.getTimeTaken()
            np.save(os.path.join(base_dir, sequence_list[i], 'nonground', '{:06d}'.format(j)), nonground)
            print(f"This is sequence{sequence_list[i]}, images{j}")
            print("Origianl Points  #: ", pointcloud.shape[0])
            print("Nonground Points #: ", nonground.shape[0])
            print("Time Taken : ", time_taken / 1000000, "(sec)")
`
However, when I check the result in cloudcompare, it seems that there is a offset between original point cloud and estimated nonground points except frame 0.

This is kitti odometry sequence 00, 000001.bin and estimated 000000.pcd
[image]

seungjae24, 2022-10-03:
Hi. Thank you for having interest on our work.

While Patchwork++ processes the point cloud, it does not add any offset to the original point cloud (I've checked it myself once more). Please check the visualized original point cloud is the input point cloud of the visualized non-ground point cloud.

Thank you :)

kang-1-2-3, 2022-10-03:
Sorry ,  I did not sort the file list.

closed 2022-10-03 by kang-1-2-3