url-kaist/patchwork-plusplus #14: confused by this line of code

wangzhuhenshuai, 2022-10-25:
Hi，thanks for  open_sourcing your amazing algorithm.I have test it in my lidar and it works well. But i am confused by this line of code:
https://github.com/url-kaist/patchwork-plusplus/blob/0d430884cbea4c3cfa009cf99a46c4f04ae964e2/patchworkpp/src/patchworkpp.cpp#L58
I have not seen any describes of this code in your paper and when i comment out this code the result seems to be the same.

seungjae24, 2022-10-25:
Hi, thank you for having interest on our work :)

Actually, this is the line for modifying the direction of the eigenvector heading to upward.

In my cases, I saw that few resultant eigenvectors with the smallest eigenvalue, which are regarded as the normal vector of the ground surface, have the downard directions.

As the following codes assume the normal vectors of ground surfaces are upward, it was necessary to make the vectors upward. So, this is the code for that.

wangzhuhenshuai, 2022-10-25:
I see it,thanks for your reply!

closed 2022-10-25 by wangzhuhenshuai