url-kaist/patchwork-plusplus #20: Temporal Ground Revert implementation clarification

lucamozza, 2023-04-11:
Hi, and thank you for open sourcing your algorithm.

By looking at the implementation of the temporal_ground_revert function it seems that the implemented version is a bit more complex than what is described in the paper.
I expected a simple threshold on this value
https://github.com/url-kaist/patchwork-plusplus/blob/11c189380b85db7f82f43e993f481aadce39b84f/patchworkpp/src/patchworkpp.cpp#L411

Could you elaborate on the implemented version and its advantages?

My guesses so far are as follows:

- A flatness probability is computed, instead of applying a "hard threshold".
- The flatness probability is manually set to 1 if the patch contains enough region-wise ground points and their flatness is below th_dist_ squared. (Is the squared value there to match units of measure or is it an empirical value?)
- A line probability is also computed, which is set to 1 if the major eigenvalue is at least 8 times the mid eigenvalue. I'm guessing the reasoning here is that a lidar ring within a patch should be close to linear on (flat ground).

seungjae24, 2023-04-11:
Hi, @lucamozza.

Thank you for having interest on our algorithm.

As you said, the detailed implementation of temporal_ground_revert function could be seemed to be slightly different from the paper. Well, those are remaining codes of our additional trial to improve reverting module's performance by considering additional geometrical values. Unfortunately, I thought other properties(e.g. linearity) except the flatness value are difficult to be concluded as useful geometric properties for estimating ground likelihood in general cases. Therefore, those are not stated in the paper.

By the way, I can answer your guesses as follows:

> A flatness probability is computed, instead of applying a "hard threshold". How is it computed?

As I commented, I was trying to improve reverting function's performance by considering additional geometric properties. You can do hard thresholding instead of calculating a probability.

> The flatness probability is manually set to 1 if the patch contains enough region-wise ground points and their flatness is below th_dist_ squared. (Is the squared value there to match units of measure or is it an empirical value?)

The flatness variable is the smallest eigenvalue from the result of PCA. Actually, the square root of eigenvalue is related to the standard deviation along the direction of eigenvector (corresponding normal vector of the surface). So, it is theoretically meaningful to compare flatness value to th_dist_squared.

> A line probability is also computed, which is set to 1 if the major eigenvalue is at least 8 times the mid eigenvalue. I'm guessing the reasoning here is that a lidar ring within a patch should be close to linear on (flat ground).

Actually, as temporal_ground_revert module operates for close enough patches, I thought it is not expected to have ground points of a single LiDAR ring, which means it is not expected to have a high linearity. The ground points in a close-enough patch are usually planar enough and not linear much. It was the reason why I tried to utilize the line probability.

lucamozza, 2023-04-12:
@seungjae24 thank you for your fast and exaustive response!
I definitely got confused while reading the line probability part. The conclusion is that we expect ground to be planar, so if we find it is more similar to a line we do NOT revert it.

I understand that the additional implemented checks do not improve the performance in general cases.
Do you suggest sticking to the paper version, or is there any difference in performance at all (in some specific cases maybe?) ?

seungjae24, 2023-04-14:
@lucamozza

> I definitely got confused while reading the line probability part. The conclusion is that we expect ground to be planar, so if we find it is more similar to a line we do NOT revert it.

Yes, the implemented part actually tried what you said. It was reasonable to revert not line-like points since reverting module is only applied for close enough patches which are containing ground points with multiple rings.

> Do you suggest sticking to the paper version, or is there any difference in performance at all (in some specific cases maybe?) ?

Those parts don't lead a dramatic change of ground segmentation performance. However, I don't think that those parts are totally meaningless to segment ground points. It might be able to improve ground segmentation based on mulitple geometrical properties if those properties are considered without loss of generality.

lucamozza, 2023-04-14:
Thank you very much for your support!

closed 2023-04-14 by lucamozza