url-kaist/patchwork-plusplus #25: I cannot run python example code (demo_sequential.py and demo_visualize). I get cannot find pypatchworkpp!

khaledmohamedgamal, 2023-05-25:
- The `pypatchworkpp.cpython-38-x86_64-linux-gnu.so` is built in `/build/python_wrapper`. However when I run  `python examples/python/demo_sequential.py`, I get this message `Cannot find pypatchworkpp!` 
- The CPP examples work fine. 
- When I was searching for similar error I found someone who solved similar problem with pybind11, and he wrote "I fixed this by making sure the module name in my bindings macro, matches the module name in the CMakeLists.txt file."
- [https://stackoverflow.com/questions/50688184/how-do-i-import-a-module-created-with-pybind11-on-ubuntu](url)

seungjae24, 2023-06-07:
Hi @khaledmohamedgamal,

Could you please try `sudo make install` after `make`? I'm not sure that it solve the problem, but please try it and let me know your progress. I will try to find other potential causes. Thank you.

khaledmohamedgamal, 2023-06-09:
Thanks a lot @seungjae24 for your reply . 
I found the solution it is related that I have many version of python in my machine, and pybind11 when creating the .so, it used  the python version in usr/local. For example, I have conda environment with python 3.9, but the python that pybind11 sees is python 3.7; therefore, when I tried to import pypatchworkpp from python script with python version different than the python version that pybind11 sees, I got this error.

Here is the solution to specify the python version:
The target Python version can be selected by setting `PYBIND11_PYTHON_VERSION` or an exact Python installation can be specified with `PYTHON_EXECUTABLE` when you run cmake. For example:

`cmake -DPYBIND11_PYTHON_VERSION**=3**.6 ..`

*# Another method:*
`cmake -DPYTHON_EXECUTABLE**=**/path/to/python ..`   

- [https://stackoverflow.com/questions/15291500/i-have-2-versions-of-python-installed-but-cmake-is-using-older-version-how-do](url)
- search for Configuration variables [https://pybind11.readthedocs.io/en/stable/compiling.html](url)

seungjae24, 2023-06-12:
Thank you for sharing your problem and solution, @khaledmohamed00!

It will help others who may struggle similar situation as you. Thank you once again :)

closed 2023-06-12 by seungjae24