url-kaist/patchwork-plusplus #26: the sensor height change?

Nothand0212, 2023-05-25:
I set the sensor height to 3.0，but on some case it will change like below?

the data is collected on boat. In some case, it work well; but some case, patchwork++ can not master.

did I need to change other params?

```
[ RNR ] Num of noises : 57
Flushed patches
[ CZM ] Divides pointcloud into the concentric zone model
=========== Temporal Ground Revert (TGR) ===========
[1, 8] mean_flatness: 0.0106572, stdev_flatness: 0.00265974
8th flat_sector_candidate / flatness: 0.0195439 / line_variable: 5.9849 / ground_num : 137
FINAL REJECT
====================================================
sensor height: 0.709483
elevation_thr_  :   -0.4289,  -0.8812,  -0.2444,  -0.1239
flatness_thr_   :   0.0152,  0.0085,  0.0013,  0.0053
Input Point Cloud Size:         18653
Ground Point Cloud Size:        5352
Non-Ground Point Cloud Size:    13301
Timer took 7.17513ms
```

seungjae24, 2023-06-07:
Hi @Nothand0212,

It is true that the sensor height is update in an adaptive manner. You can disable the update of sensor_height variable in the source code, which is written as below.

https://github.com/url-kaist/patchwork-plusplus/blob/d3f4ca4ba99970c67435627e37c5fd78a66e47ba/patchworkpp/src/patchworkpp.cpp#L338

Also, it might be great if you share your data collected using boat. It looks very interesting to try Patchwork++ in another platform because I have tested my algorithm only for autonomous vehicles, mobile robots, or quadruped robots. If you don't mind to share your data, please send me an e-mail! The e-mail address is written in the end of README. Thank you!

closed 2023-06-12 by seungjae24

GMahmoud, 2023-11-15:
Hello @seungjae24 !
Thanks for you work :) 
Can you explain to me the impact of removing such update on the sensor_height ? 
And why do we need to update it in adaptive manner ? 🤔 
Cheers 👍🏻