url-kaist/patchwork-plusplus #29: When using 16-beam lidar, there will be false detection at close range

Kazawaryu, 2023-07-12:
Thanks for your awsome work, it really helps me a lot.

But when I try to test custom data recorded by my 16-beam lidar, it happens in almost each point cloud, there will be false detection at close range, like this:
[image]

I wonder if there is any way to solve this problem, or could you please tell me the reason? Thanks a lot.

Kazawaryu, 2023-07-12:
@seungjae24

seungjae24, 2023-07-12:
Hi @Kazawaryu , thank you for sharing your situation.

Could you please check the min distance in the patchworkpp:Params? The points near than the default min distance, which is 2.7m, would be segmented as non-ground points. If you still struggle with a problem, you can share your data file then it would be better for me to understand the situation precisely. You can send a data by e-mail, it's written on the readme.

Thank you :)

Kazawaryu, 2023-07-13:
> Hi @Kazawaryu , thank you for sharing your situation.
>
> Could you please check the min distance in the patchworkpp:Params? The points near than the default min distance, which is 2.7m, would be [...]

Thank you for your help.

The data was recorded across in campus on a robot, there may be some problem using default parameters. I meticulously read your papers and change the parameters like you said, it works prefectly!

Thanks a lot for your help :)

closed 2023-07-13 by Kazawaryu