url-kaist/patchwork-plusplus #30: How to refine the performance on my own dataset?

wcshen, 2023-07-26:
Thanks for your great work and release the demo!
When i apply patchwork++ in my own data to segment ground_plane:
one obvious FP on the very flatten ground:
[image]

[image]

Also many ground-points on the obvious obstacles:
[image]

**Does it one exist problem of the pathwork++, or can i refine the performance by changing some params?**

wcshen, 2023-07-26:
This is the pcd file:
https://drive.google.com/file/d/19QOYjPbPuY2lXBslSs9FzlWYXhyOXNYL/view?usp=sharing

seungjae24, 2023-09-26:
Hi @wcshen, thank you for having interest on our work.

Patchwork++ was developed to be worked self-adaptively to reduce efforts for finding optimal parameters. However, of course, some unwanted result could be occured as you gave us. I don't think simply changing some parameters would lead a dramatic performance change in your case. Thank you.

closed 2023-10-11 by seungjae24

SuperCarryFu, 2023-10-18:
Hi! How can I modify the read_din function so that I can read the pcd file.