url-kaist/patchwork-plusplus #31: Python example

barzanisar, 2023-08-26:
Hey,

In your python example, 
```
ground      = PatchworkPLUSPLUS.getGround()
nonground   = PatchworkPLUSPLUS.getNonground()
```

These lines return ground and nonground points. How can we get their indices in the original input point cloud? I need the indices so that I can get their corresponding intensities.

Thanks in advance for your help!

LimHyungTae, 2023-08-27:
If the real-time issue is not your interest, we'd like to recommend the following directions:

1. Register raw point cloud to k-D tree 
2. Given estimated ground points, if distance btw a point from est. ground and the closest point from k-D tree is lower than epsilon (i.e. < 0.0001), label the corresponding point of raw cloud as the est. ground

LimHyungTae, 2023-08-27:
Or please refer to label generator in Patchwork v1: https://github.com/LimHyungTae/patchwork/tree/main/include%2Flabel_generator

barzanisar, 2023-08-27:
I made a [PR](https://github.com/url-kaist/patchwork-plusplus/pull/32) which stores the point index in the PointXYZ and has getter functions which simply return the indices of points.

seungjae24, 2023-09-26:
Thank you for contributing on our work. You PR will be reviewed soon!

seungjae24, 2023-10-11:
This issue has been solved. @barzanisar 's PR #32 is merged to the master branch.
If you need specific indices for ground or nonground points, please check the examples.

closed 2023-10-11 by seungjae24