url-kaist/patchwork-plusplus #34: Building on windows

rakomar, 2023-09-19:
I'm not quite used to cmake and make, but I was trying to get the examples running on my windows machine. I was using 

cmake .. -DCMAKE_TOOLCHAIN_FILE="path"\vcpkg\scripts\buildsystems\vcpkg.cmake

-- Selecting Windows SDK version 10.0.22000.0 to target Windows 10.0.19045.
CMake Deprecation Warning at build/_deps/pybind11-src/CMakeLists.txt:8 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at build/_deps/pybind11-src/tools/pybind11Tools.cmake:8 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.
Call Stack (most recent call first):
  build/_deps/pybind11-src/CMakeLists.txt:33 (include)

CMake Warning (dev) at "path"/vcpkg/scripts/buildsystems/vcpkg.cmake:859 (_find_package):
  Policy CMP0148 is not set: The FindPythonInterp and FindPythonLibs modules
  are removed.  Run "cmake --help-policy CMP0148" for policy details.  Use
  the cmake_policy command to set the policy and suppress this warning.

Call Stack (most recent call first):
  build/_deps/pybind11-src/tools/FindPythonLibsNew.cmake:60 (find_package)
  "path"/vcpkg/scripts/buildsystems/vcpkg.cmake:859 (_find_package)
  build/_deps/pybind11-src/tools/pybind11Tools.cmake:16 (find_package)
  build/_deps/pybind11-src/CMakeLists.txt:33 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- pybind11 v2.2.3
CMake Deprecation Warning at C:/Program Files (x86)/Open3D/CMake/Open3DConfig.cmake:18 (cmake_policy):
  The OLD behavior for policy CMP0072 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  "path"/vcpkg/scripts/buildsystems/vcpkg.cmake:859 (_find_package)
  CMakeLists.txt:24 (find_package)

-- Found Open3D 0.17.0
-- Configuring done (2.3s)
CMake Warning (dev) in CMakeLists.txt:
  Policy CMP0111 is not set: An imported target missing its location property
  fails during generation.  Run "cmake --help-policy CMP0111" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.

  IMPORTED_LOCATION not set for imported target "Open3D::Open3D"
  configuration "MinSizeRel".
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) in CMakeLists.txt:
  Policy CMP0111 is not set: An imported target missing its location property
  fails during generation.  Run "cmake --help-policy CMP0111" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.

  IMPORTED_LOCATION not set for imported target "Open3D::Open3D"
  configuration "RelWithDebInfo".
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) in CMakeLists.txt:
  Policy CMP0111 is not set: An imported target missing its location property
  fails during generation.  Run "cmake --help-policy CMP0111" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.

  IMPORTED_LOCATION not set for imported target "Open3D::Open3D"
  configuration "MinSizeRel".
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) in CMakeLists.txt:
  Policy CMP0111 is not set: An imported target missing its location property
  fails during generation.  Run "cmake --help-policy CMP0111" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.

  IMPORTED_LOCATION not set for imported target "Open3D::Open3D"
  configuration "RelWithDebInfo".
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done (0.1s)
-- Build files have been written to: G"path"/patchwork-plusplus/build

where "path" is my project directory.

When afterwards running msbuild I am getting: after some time those errors:

  C:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Tools\MSVC\14.34.31933\include\experimental/filesystem(30,1): fatal  error C1189: #error:  The <experimental/filesystem> header providing std::experimental::filesystem is deprecated by Microsoft and wil
_SILENCE_EXPERIMENTAL_FILESYSTEM_DEPRECATION_WARNING to suppress this error. ["path"\patchwork-plusplus\build\demo_sequential.vcxproj]

""path"\patchwork-plusplus\build\PATCHWORK.sln" (Standardziel) (1) ->
  C:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Tools\MSVC\14.34.31933\include\xmemory(154,5): error C3861: "_invalid_parameter": Bezeichner wurde nicht gefunden. ["path"\patchwork-plusplus\build\python_wrapper\pypatchwo
C:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Tools\MSVC\14.34.31933\include\xmemory(164,5): error C2039: "_invalid_parameter" ist kein Member von "`global namespace'". [G:\Projekte\FootprintPioneers\3DFoot\patchwork-plusplus\build\python_wrapper\p

where the German lines say "error C3861: "_invalid_parameter": Declerator not found." and "error C2039: "_invalid_parameter" is not a member of "`global namespace'"".

I am a bit lost with those errors or what to do about them.
Do you have an idea?

Best regards

LimHyungTae, 2023-09-25:
Hmmm...we are not familiar with the Windows environments, but we'll check it soon!

L00qqq, 2023-12-06:
You can try Windows+Visual Studio and configure the source code directly

closed 2024-03-06 by seungjae24