url-kaist/patchwork-plusplus #41: Question about sensor height parameter

GMahmoud, 2023-11-27:
Hello  !
Thanks for you work :)
This question is related to the following [issue](https://github.com/url-kaist/patchwork-plusplus/issues/26)
Can you explain to me the impact of removing such update on the sensor_height ?
And why do we need to update it in adaptive manner ? 🤔

In case we disable the update of the sensor height should we modify this condition in the noise removal process
z < -params_.sensor_height-0.8 and put z < -params_.sensor_height instead ? I did not fully understand this criteria of 0.8

[patchwork-plusplus/patchworkpp/src/patchworkpp.cpp](https://github.com/url-kaist/patchwork-plusplus/blob/d3f4ca4ba99970c67435627e37c5fd78a66e47ba/patchworkpp/src/patchworkpp.cpp#L381)

Line 381 in [d3f4ca4](https://github.com/url-kaist/patchwork-plusplus/commit/d3f4ca4ba99970c67435627e37c5fd78a66e47ba)

 if ( ver_angle_in_deg < params_.RNR_ver_angle_thr && z < -params_.sensor_height-0.8 && cloud_in.row(i)(3) < params_.RNR_intensity_thr) 
Thanks for your answer !
Cheers 👍🏻

seungjae24, 2023-11-28:
Hello @GMahmoud,

Thanks for your kind intereset :)

> Can you explain to me the impact of removing such update on the sensor_height ?

Then, the sensor height, which is a member variable and receives a value from the initial parameter setting, doesn't change.

> why do we need to update it in adaptive manner  ?

The automatically self-adaptive height estimation is desigend for 1) convenience, reducing efforts to set specific parameters, and 2) applicability for height-varying robots, such as quadruped robots. If you already measure the sensor height of your robot and plan to run the robot in the environment without any height difference, you can disable the self-update term as what I commented on #26.

> should we modify this condition in the noise removal process?

0.8 is a bit hard-coded variable. haha .. It is a threshold for removing outlier points below the actual ground points. You don't need to change the code to `z < -params_.sensor_height`.

GMahmoud, 2023-11-28:
Thank you very much for your answer. 
I will test with a static height and see if I get any improvement because I still have false positive (especially when it's raining and the ground is wet). I made the hard-coded variable and the adaption of the height configurable so I can switch easily.

closed 2023-12-05 by seungjae24