url-kaist/patchwork-plusplus #42: Whether I can skip the installation of Open3D of C++ version, if I only want to use the Python binding of Patchwork++ using pybind11?

laliwang, 2023-12-05:
While I just run the `pip install .` in patchwork-plusplus directory with running `pip install open3d` ahead, some errors interrupts as follows:

```
Processing /mnt/storage/wjh/Third_Parties/patchwork-plusplus
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... done
Requirement already satisfied: numpy>=1.23 in /home/wjh/anaconda3/envs/h2mapping/lib/python3.8/site-packages (from pypatchworkpp==0.0.1) (1.24.4)
Building wheels for collected packages: pypatchworkpp
  Building wheel for pypatchworkpp (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Building wheel for pypatchworkpp (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [45 lines of output]
      *** scikit-build-core 0.7.0 using CMake 3.27.9 (wheel)
      *** Configuring CMake...
      loading initial cache file /tmp/tmpd3hlu9k0/build/CMakeInit.txt
      -- The C compiler identification is GNU 7.5.0
      -- The CXX compiler identification is GNU 7.5.0
      -- Detecting C compiler ABI info
      -- Detecting C compiler ABI info - failed
      -- Check for working C compiler: /usr/bin/cc
      -- Check for working C compiler: /usr/bin/cc - broken
      CMake Error at /tmp/pip-build-env-vobq7vv2/normal/lib/python3.8/site-packages/cmake/data/share/cmake-3.27/Modules/CMakeTestCCompiler.cmake:67 (message):
        The C compiler
      
          "/usr/bin/cc"
      
        is not able to compile a simple test program.
      
        It fails with the following output:
      
          Change Dir: '/tmp/tmpd3hlu9k0/build/CMakeFiles/CMakeScratch/TryCompile-9QWsce'
      
          Run Build Command(s): /tmp/pip-build-env-vobq7vv2/normal/lib/python3.8/site-packages/ninja/data/bin/ninja -v cmTC_478da
          [1/2] /usr/bin/cc    -o CMakeFiles/cmTC_478da.dir/testCCompiler.c.o -c /tmp/tmpd3hlu9k0/build/CMakeFiles/CMakeScratch/TryCompile-9QWsce/testCCompiler.c
          ERROR: ld.so: object '/usr/lib/x86_64-linux-gnu/libnvidia-opencl.so' from LD_PRELOAD cannot be preloaded (wrong ELF class: ELFCLASS64): ignored.
          ERROR: ld.so: object '/lib64/libstdc++.so' from /etc/ld.so.preload cannot be preloaded (wrong ELF class: ELFCLASS64): ignored.
          [2/2] : && /usr/bin/cc   CMakeFiles/cmTC_478da.dir/testCCompiler.c.o -o cmTC_478da   && :
          FAILED: cmTC_478da
          : && /usr/bin/cc   CMakeFiles/cmTC_478da.dir/testCCompiler.c.o -o cmTC_478da   && :
          ERROR: ld.so: object '/usr/lib/x86_64-linux-gnu/libnvidia-opencl.so' from LD_PRELOAD cannot be preloaded (wrong ELF class: ELFCLASS64): ignored.
          ERROR: ld.so: object '/lib64/libstdc++.so' from /etc/ld.so.preload cannot be preloaded (wrong ELF class: ELFCLASS64): ignored.
          /usr/lib/gcc/x86_64-linux-gnu/7/../../../../x86_64-linux-gnu/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/7/liblto_plugin.so: error loading plugin: /usr/lib/gcc/x86_64-linux-gnu/7/liblto_plugin.so: wrong ELF class: ELFCLASS64
          collect2: error: ld returned 1 exit status
          ninja: build stopped: subcommand failed.
      
      
      
      
      
        CMake will not be able to correctly generate this project.
      Call Stack (most recent call first):
        CMakeLists.txt:2 (project)
      
      
      -- Configuring incomplete, errors occurred!
      
      *** CMake configuration failed
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for pypatchworkpp
Failed to build pypatchworkpp
ERROR: Could not build wheels for pypatchworkpp, which is required to install pyproject.toml-based projects
```

seungjae24, 2024-01-09:
Hi @laliwang,

We used CMAKE version with 3.25.1. You may need a higher version over than 3.20.x, as written in README :)

Thank you.

closed 2024-01-10 by seungjae24

JennySeidenschwarz, 2024-01-11:
Hi :) 
with respect to this: I'm trying to run ```pip install .``` and also installed open3D before with pip. My cmake version is 3.28.1 but I get the following error:

```
Processing /home/wiss/seidensc/Documents/project_clean_up/NTF_lidar/patchwork-plusplus
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... done
Collecting numpy>=1.23
  Using cached numpy-1.24.4-cp38-cp38-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (17.3 MB)
Building wheels for collected packages: pypatchworkpp
  Building wheel for pypatchworkpp (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Building wheel for pypatchworkpp (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [68 lines of output]
      *** scikit-build-core 0.7.1 using CMake 3.28.1 (wheel)
      *** Configuring CMake...
      loading initial cache file /tmp/tmpaqk2292d/build/CMakeInit.txt
      -- The C compiler identification is GNU 9.4.0
      -- The CXX compiler identification is GNU 9.4.0
      -- Detecting C compiler ABI info
      -- Detecting C compiler ABI info - done
      -- Check for working C compiler: /usr/bin/cc - skipped
      -- Detecting C compile features
      -- Detecting C compile features - done
      -- Detecting CXX compiler ABI info
      -- Detecting CXX compiler ABI info - done
      -- Check for working CXX compiler: /usr/bin/c++ - skipped
      -- Detecting CXX compile features
      -- Detecting CXX compile features - done
      -- Building Python wrapper
      -- Found Python3: /usr/wiss/seidensc/anaconda3/envs/SeMoLi/bin/python3 (found version "3.8.13") found components: Interpreter Development.Module
      CMake Deprecation Warning at /tmp/tmpaqk2292d/build/_deps/pybind11-src/CMakeLists.txt:8 (cmake_minimum_required):
        Compatibility with CMake < 3.5 will be removed from a future version of
        CMake.
      
        Update the VERSION argument <min> value or use a ...<max> suffix to tell
        CMake that the project does not need compatibility with older versions.
      
      
      CMake Deprecation Warning at /tmp/tmpaqk2292d/build/_deps/pybind11-src/tools/pybind11Tools.cmake:8 (cmake_minimum_required):
        Compatibility with CMake < 3.5 will be removed from a future version of
        CMake.
      
        Update the VERSION argument <min> value or use a ...<max> suffix to tell
        CMake that the project does not need compatibility with older versions.
      Call Stack (most recent call first):
        /tmp/tmpaqk2292d/build/_deps/pybind11-src/CMakeLists.txt:33 (include)
      
      
      CMake Warning (dev) at /tmp/tmpaqk2292d/build/_deps/pybind11-src/tools/FindPythonLibsNew.cmake:60 (find_package):
        Policy CMP0148 is not set: The FindPythonInterp and FindPythonLibs modules
        are removed.  Run "cmake --help-policy CMP0148" for policy details.  Use
        the cmake_policy command to set the policy and suppress this warning.
      
      Call Stack (most recent call first):
        /tmp/tmpaqk2292d/build/_deps/pybind11-src/tools/pybind11Tools.cmake:16 (find_package)
        /tmp/tmpaqk2292d/build/_deps/pybind11-src/CMakeLists.txt:33 (include)
      This warning is for project developers.  Use -Wno-dev to suppress it.
      
      -- Found PythonInterp: python3 (found version "3.8.13")
      -- Found PythonLibs: /usr/wiss/seidensc/anaconda3/envs/SeMoLi/lib/libpython3.8.so
      -- pybind11 v2.2.3
      -- Performing Test HAS_FLTO
      -- Performing Test HAS_FLTO - Success
      -- LTO enabled
      -- Building examples for c++
      CMake Error at CMakeLists.txt:41 (find_package):
        Could not find a package configuration file provided by "Open3D" with any
        of the following names:
      
          Open3DConfig.cmake
          open3d-config.cmake
      
        Add the installation prefix of "Open3D" to CMAKE_PREFIX_PATH or set
        "Open3D_DIR" to a directory containing one of the above files.  If "Open3D"
        provides a separate development package or SDK, be sure it has been
        installed.
      
      
      -- Configuring incomplete, errors occurred!
      
      *** CMake configuration failed
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for pypatchworkpp
Failed to build pypatchworkpp
ERROR: Could not build wheels for pypatchworkpp, which is required to install pyproject.toml-based projects
```

Do you have any idea?

LimHyungTae, 2024-07-28:
This issue has been resolved. Thanks for the suggestion!