url-kaist/patchwork-plusplus #43: About ground plane

huixiancheng, 2023-12-08:
Hi, Seungjae. I used the official [script](https://github.com/url-kaist/patchwork-plusplus/blob/master/examples/python/demo_visualize.py) to test Patchwork++ on example KITTI data and it performed well. I wanna know if Patch++ supports exporting ground plane equations that are similar to Ax + By + Cz + D = 0

huixiancheng, 2023-12-08:
Also, I find that the method can not cover some case. As follow, some point that above ground been seg as groud. Any hy_params to adjust this?
[image]

seungjae24, 2024-04-11:
Hi @huixiancheng,

> I wanna know if Patch++ supports exporting ground plane equations that are similar to Ax + By + Cz + D = 0

In the existing code, it doesn't have a function that exports ground plane equations. But, as Patchwork++ estimates the normal and mean of the ground points in each patch, you can get them easily.

> Any hy_params to adjust this?

Your given image is not enough for me to figure out which is the problem and how to solve it. Sorry :(

huixiancheng, 2024-04-19:
thx for your reply.  solved.

closed 2024-04-19 by huixiancheng

arun-venkat-23, 2024-06-12:
Hey @huixiancheng 

Could you please let me know on how you solved it. I am facing the same issue, there are points which are above the ground but are still categorized as ground.

wanglaotou, 2024-11-11:
> Hey @huixiancheng
> 
> Could you please let me know on how you solved it. I am facing the same issue, there are points which are above the ground but are still categorized as ground.

same issue, have you solved the problem, thanks.