url-kaist/patchwork-plusplus #45: Depth camera adaptation

yyqgood, 2024-03-07:
thank you for your work

I would like to know:
1. Can this algorithm be adapted to depth cameras?
2. If so, what should you pay attention to?

Looking forward to your reply, thank you!

seungjae24, 2024-04-11:
Hi @yyqgood!

> 1. Can this algorithm be adapted to depth cameras?

You can try Patchwork++ on point clouds from depth cameras. However, the performance of ground segmentation should be checked because point clouds from depth cameras have different characteristics from LiDAR point clouds in terms of accuracy, maximum distance, FoV.

> 2. If so, what should you pay attention to?

You should turn off the module RNR, it can be disabled by the parameter, enable_RNR. Actually, it removes undesirable reflected noises based on some properites including intensity. However, as depth cameras don't provide intensity information for points, you should disable the module.
Also, as the point cloud from depth camera have a smaller maximum range than LiDAR, you should adjust the parameters of Concentric Zone Model (e.g., number of zones/rings/sectors). You should set them to divide point clouds having an enough amount in each patch to be estimated as a ground or non-ground properly.

closed 2024-05-27 by seungjae24

elvistheyo, 2024-10-24:
> Hi @yyqgood!
>
>
> You can try Patchwork++ on point clouds from depth cameras. However, the performance of ground segmentation should be checked because point clouds from depth cameras have different [...]

> Hi @yyqgood!
>
>
> You can try Patchwork++ on point clouds from depth cameras. However, the performance of ground segmentation should be checked because point clouds from depth cameras have different [...]

how do i adjust the parameters of Concentric Zone Model in the python version?