url-kaist/patchwork-plusplus #47: Installing without sudo

YishGene, 2024-04-16:
Hi, I'm very much interested in your project, and trying to build your repo's wheel so I can try it out for myself. However, the build script is heavily dependent on root privileges/sudo. Do you have a method for building without these requirements and/or prebuilt wheels that you can share?

seungjae24, 2024-04-23:
Hi @YishGene,

Honestly, I didn't get your situtation. Could you please let me know details of your problem or sth you need?
By the way, we've tested our algorithm primarily on Ubuntu. Therefore, if you are using another environment it might be hard to get our help.

YishGene, 2024-04-23:
Thanks for replying @seungjae24, the problem is that I'm working on a system where I don't have root privileges, so I can't run commands like sudo apt-get install.

In the meantime I created a solution by building in a docker container where I have the privileges. There are a few more steps involved, if you like I can MR. 

Everything is done in Ubuntu.

seungjae24, 2024-04-23:
I see, it sounds great that you've already solved the problem by using docker container.

We're always open to get contributions. You are free to create a pull request on our repository. If you can provide the Dockerfile to create the image, properly working with Patchwork++, I would like to appreciate you.

Thank you :)

closed 2024-05-27 by seungjae24