url-kaist/patchwork-plusplus #51: Regarding varying mean sensor height

arun-venkat-23, 2024-06-12:
Hey @seungjae24 

While calculating the initial_seeds, I was wondering why does the mean height of points vary a lot. My sensor is mounted at around 1.7m from the ground but the mean height of points does go till 3.4m

LimHyungTae, 2024-06-14:
How come??? Could you see the example please?

arun-venkat-23, 2024-06-14:
Since the plane fitting is done bin-wise, there are a few bins in which the mean height of points are at around ~3.4 (trees). Due the absence of ground points in the particular region, the algorithm during the initial seeds computation takes the mean height and tags all points below the mean as ground. This is an issue, do you think there is a workaround for this?

LimHyungTae, 2024-06-14:
Then how about just applying the global z threshold before taking the point cloud as input? The points under the tree seem definite false positives, which can be easily filtered out.

arun-venkat-23, 2024-06-14:
This might work for a flat/planar ground surface. In case of slopes, this might start putting out issues

closed 2024-07-28 by LimHyungTae