url-kaist/patchwork-plusplus #53: Transformations bring mistakes in estimation

Zero-Yi, 2024-07-23:
Hi, 

thank you for this great work. But I would like to share an interesting thing:
It cannot handle the ground correctly if we transform the points from the ego frame to the world frame. 
(Left: transformed; Right: not transformed)
[image]
You can see there is a red circle in the left pic.

Is this supposed to be so, like by design we should transform the points to the ego frame first?

LimHyungTae, 2024-07-28:
It may affect the changes of z values. Btw, why don't you run ground segmentation first before transformation?

Navsea, 2025-01-21:
It looks like this is related to the Z0 zone, the zone that patchwork ignores as it assumes it will be empty.
(sensor does not look directly down, vehicle would block rays anyway)
The red circle (non ground) lies more to the rear, as you have shifted your point cloud forward (I assume you're vehicle ref frame lies behind the sensor ref frame).
You could easily confirm this by changing the minimum ring distance, see param "min_range"

closed 2025-01-21 by Zero-Yi