url-kaist/patchwork-plusplus #54: Support other sensor configurations

LimHyungTae, 2024-07-28:
Now, I've felt some guys want to test their sensor configurations. However, mostly, our focus was the SemanticKITTI dataset.

To address this issue, I tested in my Patchwork code and separate sensor configuration (see [`sensor_configs.hpp`](https://github.com/LimHyungTae/patchwork/blob/main/include/patchwork/sensor_configs.hpp) and zone model (see [`zone_models.hpp`](https://github.com/LimHyungTae/patchwork/blob/main/include/patchwork/zone_models.hpp) from the single hpp file, and it worked!

[image]

Now, we are looking for someone to help us import those codes in Patchwork to this repository.

wanghuohuo0716, 2024-10-30:
Does it support solid-state or semi-solid-state LiDARs, such as the Livox MID360, and vertical resolution angles that are not fixed (0.4 to 0.8 degrees), such as the Hesai QT128 LiDAR?

LimHyungTae, 2024-11-08:
I really want to support those sensors; do you have any idea of how to perform region-wise ground plane fitting for them?