url-kaist/patchwork-plusplus #58: elevation and flatness conditions

I3aer, 2024-10-08:
Hi!

flatness and elevation thresholds are updated iteratively. Therefore, they are set to zeros:

`   elevation_thr = {0, 0, 0, 0}; // threshold of elevation for each ring using in GLE. Those values are updated adaptively.
    flatness_thr = {0, 0, 0, 0}; // threshold of flatness for each ring using in GLE. Those values are updated adaptively. `
    
   In the `estimateGround `member function, you compare the `ground_elevation` and `ground_flatness` variables with those thresholds using the `concentric_idx`:
   `if (concentric_idx < params_.num_rings_of_interest)
      {
          is_not_elevated = ground_elevation < params_.elevation_thr[concentric_idx];
          is_flat         = ground_flatness < params_.flatness_thr[concentric_idx];
      }`
      
I have two observations:

1.       Since the z-axis is upward, the `ground_elevation` for ground planes should be always negative. Therefore,  is_not_elevated is always False. A similar problem exists, if the least significant eigenvalue of a plane `(ground_flatness)` is positive.
2. `concentric_idx` is increaded under the ring and sector loops. Then, the threshold arrays should have more than 4 values.

LimHyungTae, 2024-10-10:
At first, yes, but aren't they updated at the next step?

I3aer, 2024-10-15:
I did not check if they are updated or not as I am interested in a single scan pcd. Another question, why I get non orthogonal normals even though I set uprightness threshold to 0.99?

patch_normals
array([[-0.09144519,  0.05306055,  0.99439555],
       **[ 0.69917166, -0.604823  ,  0.3812459 ],**
       [-0.01123331,  0.04285617,  0.9990182 ],
       [-0.09687269,  0.05675028,  0.99367774],
       [-0.10806715,  0.055059  ,  0.9926178 ],
       [-0.09962063,  0.06085207,  0.99316317],
       [-0.06921859,  0.05507125,  0.9960804 ],
       [-0.06099597,  0.05220991,  0.99677163],
       [-0.04113085,  0.03393086,  0.99857754],
       [-0.09267069,  0.04970804,  0.9944553 ],
       [-0.09723388,  0.04760678,  0.9941223 ],
       [-0.07620493,  0.06242752,  0.995136  ],
       [-0.03467678,  0.0526633 ,  0.99801   ],
       [-0.04037486,  0.04701625,  0.9980778 ],
       [-0.06198691,  0.06860244,  0.9957165 ],
       [-0.08162168,  0.01962108,  0.99647033],
       [-0.07868537,  0.0243757 ,  0.99660146],
       [-0.06772294,  0.02173795,  0.9974672 ],
       [-0.00943837,  0.0398108 ,  0.9991627 ],
       [-0.01602176,  0.03286225,  0.9993314 ],
       [-0.03886676,  0.06128451,  0.9973634 ],
       [-0.07812789, -0.00297109,  0.99693906],
       [-0.0880304 ,  0.02805167,  0.99572283],
       [-0.06525222,  0.04650887,  0.9967845 ],
       **[-0.6830666 , -0.7300232 ,  0.02205181]],** dtype=float32)

LimHyungTae, 2024-10-18:
I guess those might be filtered out later after the prediction of all normals. Could you also share the exact part that you want to discuss?