url-kaist/patchwork-plusplus #60: Question about FP ground points in distant sectors

a120471, 2024-10-31:
I try out the patchwork++ recently, and thanks for your efforts on this work. It's very efficiency and user-friendly. :)

However, I encountered some FP ground points. After diving into the implementation code, I find one particular line somewhat puzzling:
https://github.com/url-kaist/patchwork-plusplus/blob/master/cpp/patchworkpp/src/patchworkpp.cpp#L268

From my understanding, when using the default parameter num_rings_of_interest = 4, all distant sector ground points seem to be consistently included in the final ground. Which means if any distant sectors missed the ground points, there is a big chance that some FP points will be included.

Is this the intended behavior? Thanks.

Navsea, 2025-01-21:
If the points are not upright, they start checking whether the sector is near or not.
**If it is not near, they say that they cannot infer any information from elevation** (and so don't need to check flatness either).
The reasoning is that the slope could be steep, so that the mean ground height could be high.
So checking the mean ground elevation and comparing that to a threshold no longer makes sense.

If you want, underneath is the calculation of how far the default parameters do look for unrealistic ground elevation.

My understanding is that the sector is assumed to be near when the concentric_idx is smaller than the num_rings_of_intrest, 4 in your case.
By default, you have multiple rings in each zone. this is defined with the following parameter:
num_rings_each_zone = {2, 4, 4, 4};         // Setting of Concentric Zone Model(CZM)

So 2 rings of Z1, and 2 rings of Z2 are considered near by default. 
The mimimum distance is: 2.7m , the max distance is: 80m! which covers a range of: 77.3m
The start of z2 is at 2.7 + 77.3 / 8 = 12.36m
The start of z3 is at 2.7 + 77.3 / 4 = 22.02m

So the near zone starts from 2.7m until 17.19m, for anything beyond that, only the uprightness is checked.

a120471, 2025-01-22:
Thanks for the reply~ It makes sense to me.
I'll dive into it later when needed.

closed 2025-01-22 by a120471