url-kaist/patchwork-plusplus #62: Question about input pointcloud

enricocovili, 2024-11-23:
Hi, first let me congratulate to you for your fantastic work.

I want to ask a simple question, just to be 100% sure: does patchworkpp support only 360° pointclouds? 
I was trying to get it to work with a partial pointcloud but i don't think that's possible

enricocovili, 2024-11-26:
Working with 180° pointcloud, my bad

enricocovili, 2024-11-29:
I recently discovered an issue with a specific pointcloud. The crashes appears immediately when processing point cloud data from an OS1 128 Ouster LiDAR. Specifically, the crash occurs when the point cloud data does not begin from grade 0 of the LiDAR sensor.
I'm wondering if there's a configuration parameter I can adjust to resolve this, or would it be more practical to preprocess the point cloud data before passing it to Patchworkpp (like changing tf or the pointcloud itself) ?
thank you

LimHyungTae, 2024-12-06:
Have you tried to run Patchwork++ on your data? I guess it does not count.

enricocovili, 2024-12-06:
> Have you tried to run Patchwork++ on your data? I guess it does not count.

Yeah, my data is the pointcloud coming from the image above. We managed to find what seems to be the real issue:

https://github.com/url-kaist/patchwork-plusplus/blob/bef40cd5d3dc8c4d5853a5c85d59884e13195e15/cpp/patchworkpp/src/patchworkpp.cpp#L489
The code crash on this line while trying to access normal_, as it has not been initialized after patchwork wasn't able to find a ground. Our current """fix""" was to change all 4 vectors (https://github.com/url-kaist/patchwork-plusplus/blob/bef40cd5d3dc8c4d5853a5c85d59884e13195e15/cpp/patchworkpp/include/patchwork/patchworkpp.h#L179) to a fixed sized in the header. Everything seems to work well with this patch.

I'm reporting this as probably, almost surely, there is a better method to accomplish this task.

LimHyungTae, 2024-12-06:
Aha, by any chance, did you change the sensor height parameter?

enricocovili, 2024-12-06:
> Aha, by any chance, did you change the sensor height parameter?

Yeah, i'm not 100% sure but i think we changed it to 1m

LimHyungTae, 2024-12-06:
But it's weird that there are none of the ground points. Could you share some pcd files for us?

enricocovili, 2024-12-06:
> But it's weird that there are none of the ground points. Could you share some pcd files for us?

In the first screenshot there were no ground point because patchwork was crashing instantly. After the changes i described earlier everything was working fine (we are still tuning some parameters)

enricocovili, 2024-12-13:
Any update on this? Can I help in some other ways?

LimHyungTae, 2024-12-18:
That's strange, though. If you have time, could you do PR about that vector one?

So you'are saying you changed the original code:

```cpp
    Eigen::VectorXf normal_;
    Eigen::VectorXf singular_values_;
    Eigen::Matrix3f cov_;
    Eigen::VectorXf pc_mean_;
```
    
to 
    
```cpp
    Eigen::Vector3f normal_;
    Eigen::Vector3f singular_values_;
    Eigen::Matrix3f cov_;
    Eigen::Vector3f pc_mean_;
```

? Do I follow you?

enricocovili, 2024-12-19:
Yeah you got the idea, i opened the PR as you asked

LimHyungTae, 2025-01-28:
Oh man I was so busy. Really sorry. I set a comment on that in #66

closed 2026-05-09 by LimHyungTae